lnls-dig/afc-hw #13: Pull-up DCDC converters' enable pins

henrique-silva, 2016-07-05:
According to the PICMG spec, when the MMC reboots is firmware, the payload must not be affected.

In the latest AFC board (v3.1), all the DCDC converters enable pins are pulled-down, so when the MMC reboots, these converters are disabled, shutting down the Payload.

aylons, 2016-08-11:
I agree that we should be able to reboot the MMC without affecting the payload.

The disadvantage to this approach would be that, when connected to a stand-alone power supply, the power supplies would be enabled by default, powering up everything if the MMC is not correctly programmed. Also, this power up would not obey any power sequence, which could pose some damage risk to the FPGA (most likely not).

How can we counter that? Could the MMC watchdog or configuration prevent the pins from changing it value during reboot?

Another option: the power enable rail is only kept pulled up while connected to the crate. In stand-alone mode, the MMC must pull some pin up to enable them.

henrique-silva, 2016-08-11:
Unfortunately the LPC1764 cannot hold its GPIOs configuration when a Reset happens, so all pins return to their original state, which is input for all the GPIOs.
Note that when in INPUT mode the pins have a "weak" pull-up, as discussed [here](https://developer.mbed.org/forum/mbed/topic/3094/)

aylons, 2016-08-24:
Maybe a transparent latch, then? So the MMC would still have to enable them beforehand, but the signal would surely be kept in place if the LPC reboots.