lnls-dig/afc-hw #15: Payload power line stays at ~2.5V when disabled

henrique-silva, 2016-07-05:
When the AMC is in the M0, M1, M2 and M3 states, the payload power is disabled and the line voltage should be 0V, but in the AFC boards it has a 2.5V tension.

It was not tested if the problem is in the AFC hardware or its a characteristic of the uTCA system (power supply, MCH, backplane).

aylons, 2016-08-11:
Where and how are you measuring it?

henrique-silva, 2016-08-23:
I've supplied the board with 3.3V in the Management Power line (bench test, not in crate) and measured 2,7V in the P12V line.
Further tests showed that the P12VAUX line is currently at 3.0V. Probably the problem is that the LDO LD1117 (IC74) is driving reserve current (from P3V3MP_AUX to P12V_AUX) and feeding the P12V line.
I've added a 8.2R resistor from P12V_AUX to GND and measured a 220mA current (probably the voltage is dropping due to the high reverse current).
This could be a problem because this is enough current to power some circuit in the Payload power line when it should be disabled.

A solution would be to add a diode from P12V_AUX to IC74, which would block a reverse current on IC74. This diode would also prevent damages to the circuit if the Stand alone power supply connector is inverted (this can happen in tests with bench power supplies).

aylons, 2016-08-24:
Besides adding a diode, we could also change the LDO for one that prevents reverse current, with the same effect.