lnls-dig/afc-hw #7: Add Payload Power (12v) flag to MMC

henrique-silva, 2016-07-05:
In AFCv3.0 and v3.1 we don't have a direct way to know if the Payload power was applied to the board. 
The IPMI standard doesn't define a specific message to let us know when we have the 12v on the board and we can't just let all the DCDC converters enabled all the time, since we have to follow a critical activation order of components in the board.
We solve this problem in the current MMC firmware by enabling only the FMC1 12v DCDC converter and reading its output through the INA220 sensor.

aylons, 2016-08-12:
Is it needed or desirable to measure the exact voltage level and stability or just checking the presence of power higher than a value is enough?

henrique-silva, 2016-08-12:
We only need to know if we have the Payload Power (+12V) applied in the channel, so any digital switching scheme should be enough. A valid threshold could be ~11.5V, since we don't use the +12v directly on the board (all power lines come from DCDCs feeded by the payload power).