lnls-dig/afc-hw #9: Change JTAG swtich from SCANSTA111 to SCANSTA112

henrique-silva, 2016-07-05:
The main feature that is only present in SCANSTA112 is the Transparent Stitcher Mode. 
In this mode, the IC can be configured via hardware to skip the addressing protocol needed, so we don't have to run a SVF configuration file on IMPACT when programming the FPGA bitstream (the JTAG mux would already be transparent a directed to the FPGA chain).

Another reason to perform this upgrade is that we can include the MMC in the JTAG chain, since the SCANSTA112 has 7 slots, rather than the 3 in SCANSTA111. This would allow us to program and debug the LPC17xx controllers using openOCD and Buspirate for example, as described in [this link](http://bgamari.github.io/posts/2012-03-28-jtag-over-buspirate.html).

The addressing pins needed to put the SCANSTA112 in the Transparent Stitcher Mode should be controlled my the MMC.

More information about these JTAG switches can be found in [this document](http://www.ti.com/lit/an/snla068c/snla068c.pdf) from Texas Instruments.

henrique-silva, 2016-07-05:
We could also remove IC73, which performs a dirty mux between FMC1 and RTM JTAG signals

aylons, 2016-08-11:
This seems like a good idea all around - remove an extra IC, make FPGA configuration easier and faster and allow for better testing of the LPC. I'm inclined to support this.

However, the STA112 is a much bigger BGA IC, which means would be placed at a packed area in the board. I just checked and the placement wouldn't be much of an issue, but the trace fanout would be.