lnls-dig/afc-hw #17: SCANSTA111 TRISTb pin must be an input

henrique-silva, 2016-07-07:
As pointed out by @abyszuk there's an error on the SCANSTA111 component packaging in the schematics. The pin TRISTb (7) is listed as an input, but according to the [datasheet](http://www.ti.com/lit/ds/symlink/scansta111.pdf) it is an output.

> **TRI-STATE NOTIFICATION OUTPUT**: This signal is asserted high when the associated TDO is TRI-STATEd. Associated means TRIST B is for TDO B , TRIST 1 is for TDO 1 , etc. This output has 12mA of drive current.

aylons, 2016-08-01:
Is this just a documentation issue, or is there any change on the actual board that must be done?

abyszuk, 2016-08-01:
It's not documentation, it's schematic bug.
Looks like designer confused TRISTb and TRSTb.
Current boards will work just fine, but for improved JTAG experience I'd propose connecting TRSTb pin to JTAG connector.
Our group at Creotech plans to do that in next revision of AFC/AFCK.

aylons, 2016-08-01:
Ok, this makes sense now.

We at LNLS are preparing a revision for bug correction and minor improvements on AFC starting next week, and this change can be done then. I intend to contact Creotech and the AFC users as soon as we check the issues in this repository, so we can coordinate the changes, avoid forks and improve the community. I'll keep you posted.

abyszuk, 2016-08-01:
OK, great. We've got a changelist on our own. It'll be great to work on this together.