lnls-dig/afc-hw #18: Level Translator TXB0104PWR

jlmais, 2016-07-11:
Level translator adopted TXB0104PWR (ic68, ic69, ic70, ic71) is unable to provide appropriate current to drive some IOs, for example I2C CIs. 

Note: Tests with TXS0104EPWR (same footprint and function as TXB0104PWR) proof the capability of it to solve the problem and substitute TXB0104PWR with no hardware modifications.

aylons, 2016-08-01:
TXS0104E will indeed solve the problem, as it is also designed for open-drain bus driving, and this is the case for I2C. The difference is the weak driving of the TXB series, which is required to allow data flow in both directions while the bus in being driven by standard push-pull circuits.

However, being open drain, the TXS0104 may hinder the ability for some RTMs with push-pull circuits to send data through these ports. TXS0104 is specified for 20Mbps while the TXB series is rated at 100Mbps.

I suggest adding the TXS series as a variation, instead of a complete substitution.

jlmais, 2016-08-02:
I agree that "TXS0104 may hinder the ability for some RTMs with push-pull circuits to send data", but I can't see hard limitations problems reducing operation rate from 100Mbps to 20Mbps because the purpose of these ports is just to control slow GPIO and serial devices.

However, I would add a AFC board manufactoring variation just in case of most demanding functionality change/option, instead of a modification driven from a specific application. The generality of these IO ports could be lost.

I suggest to maintain TXB0104 on the AFC board and evaluate to add TXS0104, or similar able to drive open-drain devices, on the RTM board. In my opinion, a issue of the application needs to be solved on the application board, the RTM board in this case.

João Leandro de Brito Neto
Beam Diagnostics Group (DIG)
Brazilian Synchrotron Light Laboratory (LNLS/CNPEM)
Campinas - Brazil - P.O.Box 6192 - ZIP Code 13083-970
Tel: +55 19 3517-5070
Fax: +55 19 3512-1004
email: joao.brito@lnls.br
www.lnls.br

aylons, 2016-08-24:
So, no change needed in the AFC board right now?